Ryzee119/usb64 #19: Use with no SD card

Ryzee119, 2020-09-28:
I noticed an issue where if you run the usb64 without an SD card, it can freeze. This shouldnt happen and is possibly a bug somewhere.

It just needs to be able to run without an SD card. This would inherently disable mempak and transferpak features. Also user settings would revert to default as they're loaded from SD card.

Other things to consider are hot plugging/removal too 🤔

Ryzee119, 2020-09-29:
Seems like it freezes if you unplug the SD card mid use then try access it (via TPAK or MEMPAK or otherwise).

It needs to be powered down and back up to get it working again

networkfusion, 2022-01-10:
It seems like the SD card is only detected on startup. This should probably be changed so that it is an event, this way the system will always know whether the SD is inserted or not.