Ryzee119/usb64 #27: Plugging in controller port causes N64 to freeze

CorcoranEd, 2020-10-22:
I thought it was originally a problem with no input to the n64 controller port as I was on the everdrive menu. It actually appears to cause the video on the N64 to freeze, because the sound keeps going in the background.

turned most of the debug flags on to see if there were any errors being generated but couldn't see anything obvious.

> 23:30:36.315 -> [FILEIO] Mounting fs
> 23:30:36.315 -> [FATFS] Initialise disk
> 23:30:36.315 -> [FATFS] Sector size 512
> 23:30:36.315 -> [FATFS] Num sectors 125141120
> 23:30:36.315 -> [FATFS] Sectors per cluster 64
> 23:30:36.315 -> [FATFS] Read from sector 0 for 1 sector(s)
> 23:30:36.315 -> [FATFS] Read from sector 2048 for 1 sector(s)
> 23:30:36.315 -> [FATFS] Read from sector 2049 for 1 sector(s)
> 23:30:36.315 -> [MEMORY] External memory initialised
> 23:30:36.315 -> [MEMORY] Detected 16MB
> 23:30:36.315 -> [MEMORY] Heap start: 70000000
> 23:30:36.315 -> [MEMORY] Heap end: 71000000
> 23:30:36.315 -> [MEMORY] Number of memory chunks: 512
> 23:30:36.315 -> [FATFS] Read from sector 32634 for 1 sector(s)
> 23:30:36.315 -> [FATFS] Read from sector 40378 for 1 sector(s)
> 23:30:36.315 -> [FATFS] Read from sector 40379 for 1 sector(s)
> 23:30:36.315 -> [FATFS] Read from sector 40380 for 1 sector(s)
> 23:30:36.315 -> [FILEIO] Reading SETTINGS.DAT for 1042 bytes ok!
> > 23:30:36.315 -> [MEMORY] Alloc'd SETTINGS.DAT, 1042 bytes at 0x20228a88
23:30:36.315 -> [MAIN] Flushed RAM to SD card as required
23:30:36.315 -> [MAIN] Flushed RAM to SD card as required
23:30:36.315 -> [MAIN] Flushed RAM to SD card as required
23:30:36.315 -> [MAIN] Flushed RAM to SD card as required
23:30:36.315 -> [INPUT] Registered gamecontroller to slot 0
23:30:51.625 -> [INPUT] Cleared device from slot 0
23:30:52.332 -> [INPUT] Registered gamecontroller to slot 0
23:30:58.427 -> [MAIN] C0 to rpak

CorcoranEd, 2020-10-22:
[image]

[image]

Ryzee119, 2020-10-23:
Thanks foe reporting this issue. I seems like its holding the data line low for some reason.

Ill have a look into it as soon as i can.

You can try pressing the button on the teensy to reset it after connecting it to the n64 and see if you can get it going.

Ive had a similar issue once before but wasnt able to repeat it

Ryzee119, 2020-10-29:
Hi @CorcoranEd were you able to solve this?

I cannot replicate the issue on my end. Can you provide details on what you are doing exactly?

CorcoranEd, 2021-09-02:
Hi @Ryzee119, sorry for the long delay, I didn't see your follow up message and put this on the shelf for a bit.

I built one of the PCBs for this on the weekend and updated to the latest release, but I am having the same issue with both my N64s. Whenever I plug in player one from the usb64 it freezes the video.

One is RGB modded and one stock, both still freeze when I plug the usb64 in. I've tried with an everdrive and with a normal game in both N64s as well and have the same issue.

[Here is a video showing the issue](https://streamable.com/c2p7a7) and some photos of my pcb. Happy to provide any extra info you might need. 

[image]

[image]

Ryzee119, 2021-09-02:
Hmm thats interesting. Does it do it if you boot with it already plugged in?

I might need to add some denouncing as theres probably a bit of mechanical noise on the initial plugin that could be confusing things

Ryzee119, 2021-09-04:
Im able to repeat the issue but only if I short the player 1 data line to ground. Can you confirm that the data line isnt shorted to ground? Maybe a dodgy cable?

Ive mimmicked your setup almost exactly (8bitdo adaptor, xboxone controller, super mario 64, three controllers plugged in and plugging in player one at the same time as you and could not get this too happen!)

CorcoranEd, 2021-09-22:
I had a look for shorts and found that data was shorted to ground and it turned out to be a short on my teensy itself. I've ordered a new one so ill let you know how I go making up a new board, but this appears to just be a hardware issue with the teensy. Thanks.

Ryzee119, 2021-09-22:
You could also redirect the input to another pin. By modifying usb64_conf.h and recompiling.

Ofcourse youd need to remove the trace to the damaged pin and wire it to the other.

Ryzee119, 2021-09-22:
Ill also add you could say swap say player 1 and player 4 then just not use player 4 atleast until a replacement arrives