Ryzee119/usb64 #39: Teensy 3.6 support

Spacejockey-GH, 2021-06-14:
Hi!
Could this work in a teensy 3.6?
Do you see that it is feasible to use the 3.3Volt of the n64 to power the teensy?

Ryzee119, 2021-06-14:
It could probably work on Teensy3.6, however due to the reduced RAM, Transferpak emulation would not be supported.
I managed to get it to compile for a Teensy3.6 quite easily, however I dont have one so cannot test.

USB requires 5V. Boosting the B64 3.3V to 5V is outside the scope of this project. I dont want to overload/damage any N64s.

Ryzee119, 2021-06-14:
See https://github.com/Ryzee119/usb64/tree/teensy36 branch for Teensy3.6 build.

Completely untested.

Ryzee119, 2021-06-29:
This build will very likely need debugging/code changes, so providing a build is not ideal and I will not be able to support this much.
Not sure what your process is but I just did this to build it:
```
git clone https://github.com/Ryzee119/usb64.git --recursive
cd usb64
git checkout teensy36
```
Then open the folder in VSCode with the PlatformIO plugin installed

Spacejockey-GH, 2021-06-30:
Thanks, I was finally able to compile and program it but it is completely dead. Tried for both ways: USB and hardwire (enable connected to GND)

In windows I do not see any serial port to try to connect via Putty. Any ideas?

Ryzee119, 2021-06-30:
Comment stuff out until you get something from the serial port, then add bits back in to identify where it's falling over