Ryzee119/usb64 #42: F-Zero X (usa) controller 1 stops responding. N64Digital

kimbapslice, 2021-07-20:
build-202012142305

USB64 + microsoft adapter + xbox360 controller/ps3+8bitdo/wired xbox360, n64digital 1.50.1/1.30.1 stable, 4MB RAM/jumper, just does not want to work with F-Zero X (usa). Game says either there's no controller or it would stop reading inputs after 20 seconds of the race. I can continue to navigate n64digital menu however. A stock n64 will work with the usb64.

Tested with original FZero X cart and everdrive 64. All other games seem to work perfectly.

So I connected 2 wireless xbox360 and started a 2 player vs in fzero x
Second controller continues to respond while controller 1 "stops" responding in game
but controller 1 can still bring up n64digital menu and soft reset shortcuts work.

I am not sure if this is an N64Digital firmware issue, n64digital install issue, or USB64 firmware related. Thanks!

Ryzee119, 2021-07-21:
Do you know if it works without the microsoft adaptor?
I have had issues missing the wireless adaptor with other controllers before.

kimbapslice, 2021-07-21:
I have tried with a wired xbox 360 controller and an 8bitdo adapter with ps3 sixaxis controller. Same result

Ryzee119, 2021-07-22:
Unfortunately I dont have a n64digital to repeat the same environment. I played it for a while 3 player with no issues.

You could tril this firmware. Ive just disabled the internal pullups incase the n64digital has pullups enabled aswell to the pull strength too high

Program it with Teensy Loader

[firmware_nopullup.zip](https://github.com/Ryzee119/usb64/files/6860693/firmware_nopullup.zip)

kimbapslice, 2021-07-22:
The no pullup firmware did not make a difference. The controller port 1 becomes in active "in-game" immediately or after about 20 to 30 seconds. Even plugging in an official n64 controller into port one does not make it active in game again. However, N64Digital menu navigation is still possible and USB64 still can read controller 1 inputs.

Only for Fzero X so far that I can tell.

kimbapslice, 2021-07-22:
Sometimes, I would see an error on the lcd screen for example:
"[N64] ERROR: Address CRC Error c00d"

I am not sure what those errors indicate or if that's from soft resetting via the N64digital back into everdrive menu. Again, I have tested this with official Fzero x cartridge as well.

Ryzee119, 2021-07-30:
Thanks for further testing. It seems like the n64digital is adding some extra capacitance/noise or something to the line. Assuming the n64digital is just passively listening.

Those errors means its reading incorrect data on the bus than expected. Address CRC mismatch (normally caused by noise).

Hopefully i can grab a n64digital in the next batch.

Make sure your grounds are solid.

kimbapslice, 2021-07-30:
Thank you Ryzee119. One other user today was able to replicate the fzero controller issue on discord.

mathieulh, 2021-10-17:
I can reproduce this on all versions of F-Zero GX, including the 64DD expansion versions.

I am using a n64digital and the Xbox 360 wireless receiver.

Ryzee119, 2021-10-18:
Unfortunately i was still unable to secure a n64digital to investigate this. Its certainly a weird issue.

Does the issue occur if only player 1 is connected? (N64 controller cables physically disconnected on the other ports?)

I have UltraHDMI without issues but UltraHDMI OSD only passively listens for button presses (with the downside of being unable to detect button presses if the game isnt actively checking controllers like on loading screens etc)

usb64 relies on the fact that the PI can only poll one controller at a time so I can easily emulate multiple controllers 'at once'.

If n64digital polls controllers itself (does anyone know if this happens?) it will break alot of things potentially as it may be in the middle of handling another controller. At this point its the only thing I can think of.

F-Zero GX may just be suseptible to it and wont recover automatically like other games. Most games may not care about of a few corrupt packets.

mathieulh, 2021-10-18:
The N64digital behaves exactly like the ultrahdmi and only passively listens to inputs. The inputs aren't recognized while games aren't actively checking for controllers

Ryzee119, 2021-11-01:
Hopefully ive secured a N64 Digital to investigate/fix this issue.

I put my scope on the joybus lines while playing FZero and there's not much happening that should be a challenge for usb64.
You could try the latest FW which has a few fixes but nothing that Im confident would fix this issue, but hopefully I get my unit in a few weeks to work this out.

mathieulh, 2021-11-03:
The latest firmware still has the same issue.

networkfusion, 2022-01-10:
The N64 digital uses JoyBus for Game ID detection. It is likely this is causing un-intended side effects. If using an ED64 V3 or lower, you could try my unofficial OS https://github.com/N64-tools/ED64-UnofficialOS-binaries which allows you to "disable" the joybus send.

For further investigation, this is the code used to send the game id: https://gitlab.com/pixelfx-public/n64-game-id

mathieulh, 2022-01-10:
> The N64 digital uses JoyBus for Game ID detection. It is likely this is causing un-intended side effects. If using an ED64 V3 or lower, you could try my unofficial OS https://github.com/N64-tools/ED64-UnofficialOS-binaries which allows you to "disable" the joybus send, or even use an older firmware on an X7 or X5 pre N64Digital support.
> 
> For further investigation, this is the code used to send the game id: https://gitlab.com/pixelfx-public/n64-game-id

Unfortunately I am using an x7 everdrive so I can't test this.

networkfusion, 2022-01-10:
> Unfortunately I am using an x7 everdrive so I can't test this.

I updated the comment... or even use an older firmware on an X7 or X5 pre N64Digital support.

Ryzee119, 2022-01-10:
@networkfusion nice thanks!

Didnt know about this. That would certainly explain the issue.

Can everyone confirm they're using an everdrive x5/x7?

If thats it, it doesn't seem too hard to fix

kimbapslice, 2022-01-10:
@networkfusion Ok, I just tested your V2.12.10.Preview-2 build on an ED64 v2.00, n64digital 1.6.6, latest usb64 build-202110290732. With and without the n64digital option in the ED64 menu turned on/off, the controller stops responding still after some time.

I couldn't find the joybus option or did I misinterpret what you meant. Sorry I do not have a x-series cart to test.

networkfusion, 2022-01-10:
> 

No miss interpretation. If the `N64 Digital` option was `off` in my unofficial menu, it means that the `joybus` commands were not sent from the `ED64 Cart`. So... it possibily becomes a noise issue 😓

Ryzee119, 2022-01-10:
I thought I managed to secure a n64Digital but still waiting for the next batch now :(

networkfusion, 2022-01-10:
> I thought I managed to secure a n64Digital but still waiting for the next batch now :(

Once we get the F7 series working, and I can find a soldering iron, I can definely add some input here...

kimbapslice, 2022-01-10:
With your version networkfusion, all of the games I tried show the gameid name in the n64digital menu correctly. FZero X (USA) does not pass the gameid - it's just blank.  I actually used a retrode and ripped the rom of my original cart md5 verified 753437d0d8ada1d12f3f9cf0f0a5171f and the gameid was still blank.

Running the original fzero x cart and just leaving it on the main game menu, after a few seconds would display "no controller". Anyway, not a big issue, and just this one game to my experience.

I hope Ryzee119 will be able to get a n64digital soon, but it's like catching a unicorn at the moment.

SubElement, 2022-03-20:
Unsure if this helps in any way, but I just tried this with Nintendo Switch Online N64 Controller + 8bitdo Adapter and barring a separate button mapping issue this is working totally fine for me on my N64 with an N64 Digital.

Let me know if I can help further with testing.

nmcgill, 2022-04-10:
I experienced this as well in F-Zero, one difference here is I am using UltraHDMI. Here is my setup:

UltraHDMI HW1 (firmware v1.10)
8bitdo Adapter (firmware v2.03)
Nintendo Switch Online N64 Controller
[EverDrive 64 V3](https://krikzz.com/our-products/legacy/everdrive-64-v3.html)

I haven't gone further in to debugging yet

darthcloud, 2022-05-20:
@Ryzee119 A similar issue  with N64 Digital/F-Zero with BlueRetro was reported to be fixed when I changed the data line GPIO config from open-drain to push-pull (regular) if that help. Not sure if it applies to USB64.

Ryzee119, 2022-05-21:
@darthcloud hey thanks for the idea. Love your work with blueretro too :D

If someone wants to try this is let me know if it fixes the issue that would be great!
[firmware-pp.zip](https://github.com/Ryzee119/usb64/files/8746059/firmware-pp.zip)
 @kimbapslice

kimbapslice, 2022-05-21:
> @darthcloud hey thanks for the idea. Love your work with blueretro too :D
> 
> If someone wants to try this is let me know if it fixes the issue that would be great! [firmware-pp.zip](https://github.com/Ryzee119/usb64/files/8746059/firmware-pp.zip) @kimbapslice

I just tested this version with my official xbox 360 wireless controller and microsoft adapter. I noticed that the rumble would be hit and miss and sometimes would constantly rumble (stuck) and then no rumble no matter what. Then about 1 minute into the race it would rumble constantly and then the in-game control would become unresponsive. I was still able to get into the n64digital menu and navigate it.

Ryzee119, 2022-06-03:
@kimbapslice damn thank for trying. Im still trying to get ahold of one.
My attack plan would be to trying an external pullup resistor on the data lines (maybe 2k or so?) and also get a oscillope on the lines to see whats going on.
I could experiment with software based glitch filtering but would be hard without being able to replicate the issue locally.

kimbapslice, 2022-06-03:
> @kimbapslice damn thank for trying. Im still trying to get ahold of one. My attack plan would be to trying an external pullup resistor on the data lines (maybe 2k or so?) and also get a oscillope on [...]

No worries, there are lots of games to play :) Thanks for your continuous work to resolve the issue!

Ryzee119, 2022-07-31:
[image]
I got one :) and I am able to repeat the issue.

Seems like its just some high frequency noise. The act of me simply putting my oscillope probes on the joybus fixes it. I may be able to fix in software by tweaking the slew rates, or worst case adding a small capacity on the data lines (~10nF or so) but will investigate

Ryzee119, 2022-07-31:
Unfortunately the GPIO is already set at the most forgiving slew rates and drive strengths.
I was able to fix it by adding a 10pF cap between Player 1 data line and gnd

[image]

OR

[image]

kimbapslice, 2022-07-31:
Looks like a clean fix with the capacitor route. I will try it when I get back home. Thank you and congrats on finally getting a n64digital :)

Ryzee119, 2022-07-31:
Although it appears to only be visible in Fzero, This cap is probably a good idea for any game..
I think its just most noticable as this game just gives up if it misses a few packets, other games must keep retrying so its less noticable.
It must be that PIF wire flapping in the breeze as part of the n64digital picking up some noise. Why it doesnt happen with UltraHDMI, im not sure (maybe the UltraHDMI FPGA has higher internal capacitance?)

nmcgill, 2022-07-31:
I've experienced this issue before using UltraHDMI (only once and haven't
repeated it since). Would the additional capacitor be recommended in this
case as well?

Ryzee119, 2022-07-31:
> I've experienced this issue before using UltraHDMI (only once and haven't repeated it since). Would the additional capacitor be recommended in this case as well?

I may have just gotten lucky with my UltraHDMI and noise issues. I would recommend this cap for UltraHDMI too.

darthcloud, 2022-07-31:
@Ryzee119 nice solution, technically much better than what I did, I'll try to see if same fix work for BR.

mathieulh, 2022-08-03:
I did the fix adding 0402 (that's what I had on hands) 10 pf 50V capacitors between the data and GND pins of the usb64 board on each of the 4 controllers? (I mean, why should we only fix controller1? It's likely noise tolerance isn't that good to begin with even without the PIF wire or this issue wouldn't have happened to begin with) and it works great. I could finally finish a cup in F-Zero X.

I would suggest you add this fix as part of the original wiring instructions.

Ryzee119, 2022-08-03:
Thanks for testing! Having this cap for all for controllers certainly wouldn't hurt.
I'll update the breakout PCB and the relevant readme sections to capture this asap

kimbapslice, 2022-08-06:
Hi Ryzee119,

I was able to finally install the 10pF capacitor on the main board between ground and pin 36. F-Zero X is working great now :) Please let me know and I will close the ticket. Thanks for the resolution.

kimbapslice, 2022-08-06:
I have added a 10pF capacitor (ceramic, yellow) from ground to pin 36 on teensy board but it still shows "No controller" after I leave the game on idle for a while. USB64 firmware from October 29th, 2021, n64digital firmware 2.0.0 (beta). I was able to finish a race, but main screen idle still an issue. So weird.

Ryzee119, 2022-08-06:
The 10pf isn't an exact science. Although I found 1nF too big.
If you have any other values handy you could try 100pF or something. 
You could also try adding another 10pf in parallel with the existing to get 20pF

kimbapslice, 2022-08-08:
Ryzee, I removed the 10pF and soldered on the 100pF and so far it is working great!

Ryzee119, 2022-08-08:
Ive updated the relevant HW  documentation in commit https://github.com/Ryzee119/usb64/commit/f60e9b89e010cf2742cf1a59cd9bd069aef05636.

Also for better visibilty ive created a new pinned issue highlighting the fix we have found. https://github.com/Ryzee119/usb64/issues/59.

I will close this here, but we can continue to discuss here and we can reopen if needed.

closed 2022-08-08 by Ryzee119

darthcloud, 2022-08-13:
For BlueRetro adding a 10pF or 100pF had no effect, what did the trick was adding a buffer gate (See https://github.com/darthcloud/BlueRetro/issues/264#issuecomment-1214224340)

Maybe that might help USB64 as well if the cap don't work.

[image]