Ryzee119/usb64 #43: Controller (Titan One running in Xbox 360 output mode) hangs USB64 USB host behind USB hub

IonBlade2K, 2021-07-30:
I'm using a [Titan One adapter](https://www.consoletuner.com/products/titan-one/) between my Xbox One controller and the USB64, set to output to the USB64 as a Wired Xbox 360 controller.  This lets me perform remapping on an Xbox One controller before it hits the USB64 - for example, I have a script that lets Back+Y toggle between standard controls, swapping the two sticks, and enabling the analog stick to send D-pad buttons, for various games.

This works perfectly when plugged directly into the USB64.  However, when I plug it in behind a hub (I've tried 4, one of which is wall-powered to rule out power related issues), the USB64 stops responding to further controller plug / unplug events, and stops passing or handling any input, _except_ in one odd edge case where plugging in an Xbox One controller into the hub first, _then_ plugging in any number of Titan Ones into the hub allows them all to work as expected.

I compiled the firmware with #define DEBUG_USBHOST 1, though this doesn't seem to have provided a lot of extra information.  I have included all steps to recreate the issue, as well as the results I get on the TFT and from the Serial log with that flag set (serial results shown in code formatting) at each step are below, to demonstrate the issue.  I've tested each of these tests 5 times to validate the issue is reproducible in my setup.

### Test 1: Demonstrates that USB64 does not stop responding to input / control unplug and plug events when Titan One plugged in directly, but does when Titan One plugged in through USB hub 
Starting state: nothing plugged into USB64 USB port, USB64 plugged into controller ports 1-4 on N64 and USB power, Titan One programmed to output in Xbox 360 mode.

1) Plug in Titan One to USB port on USB64 directly (works as expected).
    ```
    JoystickController::claim joystickType_ 5
    [INPUT] Registered gamecontroller to slot 0
    Xbox 360 Wired: Led Status 00
    Xbox 360 Wired: Rumble Status 03
    ```

    TFT: updates to show a Rumble Pak on controller port 1.

    Behavior: Can use controller as expected.

2) Unplug Titan One from USB port on USB64 (works as expected):
    ```
    [INPUT] Cleared device from slot 0
    ````

    TFT: updates to show nothing connected to controller port 1.

3) Repeat steps 1-2 to show that this issue isn't caused by hotplug USB (works as expected).

    Behavior: Same output, controller still works as expected afterwards.

4) Plug in USB hub to the USB64, with nothing plugged into the hub.  

    Behavior: No output is logged, no changes to TFT, as expected, since no controller is plugged in yet.

5) Plug Titan One into first port on USB hub (issue begins):
    ```
    JoystickController::claim joystickType_ 5
    [INPUT] Registered gamecontroller to slot 0
    ```

    TFT: Shows one controller with a Rumble pak inserted, as expected.

    Behavior: 
        - USB64 does not respond to controller input itself (e.g. LB + RB does not change the TFT to the info page)
        - USB64 does not pass through controller inputs to the N64
        - Unplugging the Titan One from the USB hub, plugging in more controllers to the USB hub, or unplugging the USB hub from the USB64 altogether do not cause any more log entries on the COM port logging, nor does the TFT update to reflect the removed / added controllers.
        - The only sign the USB64 is still alive is that turning off the N64 power will cause the TFT to change in the upper right from "N64 is ON" to "N64 is OFF."  So the USB64 is not totally crashed, but it's effectively hung as far as USB input handling goes, until I pull power from the USB64 and plug it back in without any controllers attached.

    If I power up the USB64 when the USB hub is plugged in and the Titan One is already plugged into the hub, I get the same behavior as in step 5, with the same last line of the log before the log / TFT stop responding to controller plug / unplug / input events.

### Test 2: Demonstrates a workaround case in which a Titan One does work behind a USB hub with USB64, if an Xbox One wired controller is plugged into the USB64 before plugging into the Titan One

I'm hoping this test can lead to some idea as to where the problem might be located and how it could be fixed, as it shows the Titan One actually works perfectly fine with the USB64 behind a USB hub, and that the core issue seems to be that something is hanging, seemingly on initialization, that doesn't occur if an Xbox One controller is plugged in to the hub first.

Starting state: nothing plugged into USB64 USB port, USB64 plugged into controller ports 1-4 on N64 and USB power, 2x Titan One programmed to output in Xbox 360 mode (test can also be performed with just 1x Titan One, though end desired state is 2x controllers hooked up through Titan One, through USB hub, to USB64).

1) Plug a wired Xbox One controller into the hub (works as expected).
    ```
    JoystickController::claim joystickType_ 3
    [INPUT] Registered gamecontroller to slot 0
    ```

    TFT: updates to show a Rumble Pak on controller port 1.

    Behavior: Can use wired Xbox One controller as expected.

2) Plug a Titan One into another port on the hub (works as expected).
    ```
[... 28 lines cut ...]

    Workaround: At this point, plug USB64 outputs 2 and 3 into N64 controller inputs 1 and 2 respectively, and the first Titan One will be fed into N64 controller port 1, while the second is fed into N64 controller port 2, allowing 2 player play with Titan One devices in the mix, without the USB64 immediately crashing on plugging them in.

5) (Without having swapped the controller output plugs around per the workaround) Plug the wired N64 controller back into the hub (issue begins).

    Behavior: Nothing is logged to the serial log, and the TFT does not change to reflect the previously working wired Xbox One controller that was plugged into the hub before.  No controllers pass input to the N64, and the USB64 no longer responds to shortcuts.  We have effectively entered the state from test 1 step 5 again.

### Additional Information
1) These are the hubs I have tried (tests above were performed using the first hub):
    - Externally powered Belkin USB 2.0 hub model F5U234 Rev. 3
    - Unpowered Belkin travel USB 2.0 hub model F4U015-NPS
    - Unpowered Unitek USB 3.0 hub model Y-3045
    - Unpowered Targus travel USB 2.0 hub model ACH113US

2) If I plug the same hub with 1x Titan One, 2x Titan Ones, or 2x Titan Ones and an Xbox One wired controller into my PC instead of the USB64 as in the above tests, I do see the expected controllers (either 1x Xbox 360 controller, 2x Xbox 360 controllers, or 2x Xbox 360 controllers and an Xbox One controller, respectively).  All 3 controllers work as expected in Windows, and report the same VID / PID as an original controller (for the Titan Ones outputting in Xbox 360 output format, they report VID_0453 and PID_028E, and Windows sees them as a genuine, original Xbox 360 controller).

3) I do not own any original Xbox 360 controllers to redo these tests with, but do have Xbox One controllers.  I have performed the tests above with 3x  wired Xbox One controllers, and do not get similar issues.  These only seem to occur with my Titan One in Xbox 360 output format.

4) Given how the Titan One is able to even pass PS4 / PS5 / Xbox One / XSX auth and work with the controller accessories such as mics and such through controllers connected to it when on those original systems, its protocol handling seems to be generally very accurate; it's possible the same behavior could be seen with original Xbox 360 controllers.  I just don't have any original 360 controllers to validate.

5) Just to preemptively answer the obvious question: why not just buy 8bitdo adapters?  2 reasons:

    - The 8bitdo doesn't seem to do any sort of remapping.  I use the scripting functionality in the Titan One to allow some games with janky control schemes to play better, e.g. allowing more modern controls in Quake 2 by swapping the c buttons to the left stick for move / strafe, and the analog input to the right stick for analog look, or putting the D-pad on the analog stick for games that have d-pad only input.

    - I have a set of N64 OEM controllers that have been adapted over to RF wireless between a transmitter that slots into the rumble pak slot and a receiver wired to an N64 controller cable plug.  The circuit used to make them RF doesn't support memory pak saving, so prior to USB6 [... cut ...]  -> N64 controller female to USB HID adapter -> Titan One (converts USB HID input to Xbox 360 output, and lets me write macros to trigger USB64 back + button functionality through alternate combos on the N64 controller, since the original N64 controller doesn't have a back button) -> USB64 (which enables virtual memory pak usage with the wireless-adapted controllers).  It's the only solution I've found to let me play N64 games on an OEM controller, wirelessly, with memory pak save functionality.

IonBlade2K, 2021-07-30:
Update: I did some more digging and found that there's an additional line that could be uncommented in USHHost_t36\USBHost_t36.h (line 60 - "//#define USBHOST_PRINT_DEBUG") to print significantly more USB host debugging information.  Using a build with that uncommented, I was able to get logs for what's happening internally in the USB host code in tests 1 and 2 above.
 
[Test1 - USB Host Logging.txt](https://github.com/Ryzee119/usb64/files/6910297/Test1.-.USB.Host.Logging.txt)

[Test 2 - USB Host Logging.txt](https://github.com/Ryzee119/usb64/files/6910314/Test.2.-.USB.Host.Logging.txt)

The key takeaway here is that every time in the tests outlined above that the USB64 goes into a state that the input stops responding (either through to the N64 or on the USB64 itself, e.g. LB+RB to change pages), the hang is immediately preceded by the serial output line "ERROR Followup."

Searching for that in the code took me to USBHost_t36's ehci.cpp.  I did some digging on the EHCI implementation in USBHost on the PJRC forums and found [a thread with someone having a very similar issue with a MIDI device](https://forum.pjrc.com/threads/62787-MIDI-device-stops-working-when-connected-through-hub-(using-USBHost_t36-library)?highlight=hub+hang) - it works directly plugged in with USBHost_t36, but not through 3 different hubs.  

In that thread, Paul Stoffregen suggested to uncomment this line to force the USB Host to run at 12 Mbps (USB Full Speed, instead of 480 Mbps Hi Speed):

 `//USBHS_PORTSC1 |= USBHS_PORTSC_PFSC; // force 12 Mbit/sec`

I compiled and flashed a build with this change, and suddenly the Titan One works as expected behind a USB hub - I can use just one, or two, without needing to first plug in an Xbox One controller as in the previously mentioned workaround.

At this point, the only issue I'm still facing is that if I boot with a USB hub plugged in, with more than one TItan One is plugged in at boot time, the USB64 still goes into a hung input state, not responding to controller inputs or plugs / unplugs.  So for now, I'll just have to keep one Titan One plugged in at boot, and only plug the second in after boot when planning to play a multiplayer game.  That one is really not that big a deal anyway, since there doesn't seem to be a way currently to say "the controller in hub port x is always N64 controller port #y" so booting with more than one Titan One plugged into the hub would just mean I wouldn't know which controller is #1 - #4 without testing them all anyway, so it's almost easier to boot with just one controller connected and then plug them in one by one anyway.

I'll leave the thread open just in case someone wants to chime in with any additional info, as it would be interesting to know if anyone with 2+ Xbox 360 wired controllers could try plugging them in behind a hub and see if they get the same behavior as my tests outlined above on the current code or not, to know if this is something specific to the Titan One's Xbox 360 output mode not quite matching a real Xbox 360 wired controller.  If the behavior I'm seeing in the test above extends to real wired Xbox 360 controllers as well when plugged in behind a hub, it might make sense for USB64 to make official 12 Mbps USB releases.  If it doesn't, then with each new release, I'll just compile my own personal build with 12 Mbps forced for my setup, no big deal!

Ryzee119, 2021-07-31:
Awesome. Thanks for investigating that.
I'll certifiai