Ryzee119/usb64 #64: NSO n64 controller disconnects when trying to switch paks

supavgx, 2022-10-26:
As title says using a Japanese release of the NSO n64 wireless controllers connected to 8bitdo usb dongle v2 i used [usb64_8bitdo_n64fix.zip](https://github.com/Ryzee119/usb64/files/8319402/usb64_8bitdo_n64fix.zip) from #57 and rumble and controller works great however i cant switch to any other pak.
Quickly Pressing button combos below does nothing at all and pressing and holding down any combo rom list then disconnects me. LCD still displays Rumble Pak on first page and on second page
LCD displays cleared controller from port 0 then quickly displays registered controller from slot 0 but cobntroller then doesnt have input on game. Only way to fix so far is to pair a new controller to the dongle as initial paired controller cant gain back input control even after power reset.

Can we use zr home or snap buttons as back instead of Z or can we use those unused buttons to switch paks simply by pressing one zr=rumble,  home transfer pak etc

Ryzee119, 2022-11-23:
Can apologies for the delay, missed this one.

Some users have reported issues with HDMI modded consoles (although particularly N64Digital) See https://github.com/Ryzee119/usb64/issues/59

But actually that build is off dev branch which has loads of development changes so that may be biting us. Does normal release work https://github.com/Ryzee119/usb64/releases/tag/build-202210260119. Id expect it to work normally however the button mapping will be wrong.

supavgx, 2022-11-25:
I did see #59 but mine doesnt disconnect unless pressing down on button combo. I tried  https://github.com/Ryzee119/usb64/releases/tag/build-202210260119 and it does work all buttons seem to work fine except changing paks still not working i try all combinations and none do anything also nothing shows up on second page of  usb64 lcd so I know its not registering. at least the disconnecting issue is gone if i hold it down nothing happens but doesnt disconnect using this release. Can you help me fix please, Can you make it so that ZR acts as back right now ZR acts as B I think

supavgx, 2023-09-04:
yes it work but your right button mapping is wrong, canyou ell me how to fix please?