Ryzee119/usb64 #65: Question - 8BitDo USB Wireless Adapter 2

ThisIsHP64, 2022-11-28:
Is USB64 compatible with the [8BitDo USB Wireless Adapter 2](https://www.8bitdo.com/wireless-usb-adapter-2/)? The README only mentions compatibility with [an older version of the adapter](https://www.8bitdo.com/wireless-usb-adapter/).

I ask because USB64 seems to be the only realistic way I can use a Nintendo 64, as I have a disability and cannot grasp an original N64 controller or controllers such as the Hori Mini Pad. I can, however, use a wireless Xbox Series controller with decent dexterity.

supavgx, 2022-12-01:
es read back on #57 and #64  i use two version 2 dongle but having issues using all options

ThisIsHP64, 2023-01-15:
Bit the bullet and bought all the components, it works just fine with this adapter and the Xbox Series controller.

Thank you, Ryzee, for giving me the opportunity to finally be able to play my favorite games. I am overjoyed!

closed 2023-01-15 by ThisIsHP64